victorkemp/Mecway #1: API function for Refine x3

victorkemp, 2023-06-30:
Context: https://mecway.com/forum/discussion/comment/7137/#Comment_7137

german-bresciano, 2023-07-02:
In order of a Refinex3 and other possible meshing tools to be more useful, it would be good to have a function to **select a list of element, faces or nodes.**
I suppose it can be done by creating a new named selection, add the list of items to it, use it and then delete the named selection, but this is not efficient.

victorkemp, 2023-07-20:
You mean pausing the script and returning to the GUI for the user to select with the mouse? That would certainly be useful but it's technically quite challenging so I haven't done it.

Anyway, version 21 will have Refine x3 and other refinement tools in the API.

german-bresciano, 2023-07-20:
No, I mean to make a list in Python an pass it to a función that makes that list the active selection. Then the refx3 función (or any other function) refine that selection. 
All without the GUI. 
It may be useful to use with other future funciona that do something with the active selection.

victorkemp, 2023-07-20:
I see. What I'm doing instead with meshing tools is you pass the list of nodes/faces/elements directly to the function without using the selection. So I think selecting things won't be needed for that?

german-bresciano, 2023-07-20:
That would be perfect.

victorkemp, 2023-07-20:
Will be in version 21.

closed 2023-07-20 by victorkemp