victorkemp/Mecway #8: Named tables

german-bresciano, 2023-07-02:
It would be good if tables may be given different names and also having a checkbox for "use tablename.csv as  file name" in the table dialog.

SergioP1975, 2023-07-03:
Also a selector in the table dialog to select saved tables. Very often they are used for several loads.

victorkemp, 2023-08-21:
Version 22 will have named tables.

@SergioP1975 can you clarify what you mean by "Also a selector in the table dialog to select saved tables. Very often they are used for several loads."?

@german-bresciano  I'm still thinking about the automatic filename. So far, table names are going to be optional so using them for the filename would be odd if there isn't a name. Perhaps table names should be required (defaulting to Table, Table(2), etc.) and have an option for automatic filename which is <table name>_<configuration name>.csv if there are multiple configurations or just <table name>.csv if there's only one?

german-bresciano, 2023-08-22:
That would be great.

SergioP1975, 2023-08-22:
@victorkemp Imagine the case that you define a complex amplitude and want to use in differente loads, one could then re-use the same amplitude.

victorkemp, 2023-08-23:
@SergioP1975  Oh, looks like you're talking about the tables for defining time-dependent loads, and German is referring to the tables for showing solution data. I'll make a separate issue for that.

victorkemp, 2023-08-25:
Rather than a fixed filename option, I've added two new symbols you can put in filenames - \<t\> which is replaced by the table's name and \<c\> which is replaced by the configuration name. The latter allows you to use a single table to output a separate file for each configuration. This will be in v22.

closed 2023-08-25 by victorkemp