victorkemp/Mecway #9: Improvements on graphic representation of BC (color, size, density)

SergioP1975, 2023-07-03:
Hi, would be greate to be able to change the color, size and density of boundary conditions for better reporting. In this model I have applied two different pressures, but is not possible to identify each one. What about a switch in the Options menu to select  "one bc graphic representation on every node/element o surface" or "only a few of them because I want to see the geometry as well"

[image]

victorkemp, 2023-07-04:
Perhaps "only a few" should be  the only option? Maybe just at a few points around the edges of the selection.

Having them on every face/node was just because it was easy to implement that way.

SergioP1975, 2023-07-04:
I would preffer only a few, now I felt that the model is too crowed of widgets (and I always finish hiding it for showing the BC, that is contradiction). About showing it only in the edge of the selection don´t know if is the right choice, most of softwares shows severals markers in the middle of the selection. Well, thinking while I writhe, a combination of both could be interesting, because it will show the delimitations of the selection (something hard to tell when there are few markers).

What I think is that only in the edges will trigger a lot of questions like ... the pressure is only applied in the border???