victorkemp/Mecway #10: Add a mechanism to change de geometry reference (step file name)

SergioP1975, 2023-07-07:
Hi Victor, could you add a new option in the contextual menu of the imported parts to change it? Now we can force a reload, but not a change in the file within the Mecway GUI (I´m doing it now editing the liml file every time that I update my qeometry and want to keep the older)

victorkemp, 2023-07-08:
For now, you can remove the file, then a new item "Change file location" appears at the bottom of the context menu but you have to enter the path instead of clicking.

SergioP1975, 2023-07-10:
Thanks, then the procedure would be

Import as "part.stp", and solve
Rename part.stp as part-rev00.stp
Export from CAD the new revision as "part.stp"

Maybe with this procedure Mecway will be happy, but from outside how one can easily understand that part.stp is a more new desing than part-rev00.stp? Imagine now that you are seeing the files one or two years ahead, that's why I preffer to start with a name such as part-rev00.stp and every new model the revision level changes, so I do it manually in the liml file.

victorkemp, 2023-07-10:
Sure. This is just a workaround. I agree it should have the option to change it directly.

victorkemp, 2023-08-07:
In v22, the option to change the file will be available even when it's not missing and it will have a proper file browser.

closed 2023-08-07 by victorkemp