victorkemp/Mecway #17: Improvements in the time line plot

SergioP1975, 2023-08-23:
Hi Victor, is there a possibility to add some new features to the time line plot?

• Plot several points?
• Manual min/max definition of vertical scales?
• Adding a second vertical scale to a different variable?
• Horizontal/Vertical division lines
• X/Y Ploting (reaction load vs deflection curves)
• Attachable to the vertical side of the program window?

SergioP1975, 2023-10-04:
> I would also like to request updating the requested node value when moving the time slidebar. (VM in the example picture)
> 
> [image]

What a nice curve! What plastic material are you using?

victorkemp, 2023-10-04:
Thanks for all the suggestions Disla and Sergio. 

Please try to keep separate ideas that make sense in isolation in separate Github issues so they can be closed when finished.

victorkemp, 2024-06-19:
> • Plot several points at the same time (maybe selecting the nodes with CTRL)? 

Could you explain a bit about how you would use this? I'm thinking that the graph could display a selectable number of points and each time you click one, it add it, wrapping around to replace the older ones if you exceed the limit.

SergioP1975, 2024-06-19:
> > • Plot several points at the same time (maybe selecting the nodes with CTRL)?
> 
> Could you explain a bit about how you would use this? I'm thinking that the graph could display a selectable number of nodes and each time you click one, it add it, wrapping around to replace the older ones if you exceed the limit.

I would love to be able to select a node, and then the node gets a round visible mark in the graphic view (just like the blue one in the above pictures), and then the result gets ploted along time. Then pressing CTRL, a second, third... (I don´t know, maybe 10 more?) could be also selected, but now every other node gets a different color bubble, that match the color line in the plot, so the user could easily identify which nodes belong to which curve in the plot. This color bubbles should persist even if the user change the display mode to surfaces or elements.

One thing that happen actualy, is that we can select and plot the node very easily, but there is no direct way to show which node was selected for ploting when we change the display mode to surface or elements (because the red dots of the nodes are a little disturbing, not for picking but for reporting when the mesh is too dense)

victorkemp, 2026-10-17:
Version 35 will have:
- Multiple graphs on the timeline
- Persistent colored markers on nodes to show which curve they're associated with.
- Using the 2nd Y axis for quantities with different dimensions.
- An option to put the timeline on the right of the windows instead of the bottom.

closed 2026-10-17 by victorkemp