victorkemp/Mecway #43: Edge detection tool doesn´t work well with too thin wedge/hexa elements

SergioP1975, 2024-05-30:
Hi, I have noticied that for elements too thin the Edge detection tool very often fails to detect the edge of the element in the "corner" of the shape and stop the selection, even with a 0° thresshold:

In this example (a 2d second order quad mesh that was revoluted 1° around Y axis with 1 element in thickness), a face in the front is selected, but the selection turns the corner as can be seen in the second picture.

Also, if I select one face on the oposite side, with thresshold 0°, not all the faces get selected

[image]

[image]

Regards

victorkemp, 2024-05-30:
The element on the edge might be a collapsed hex instead of a wedge. I see the selection goes around the sharp corner in that case. If that's the problem, this issue is more that revolve should convert those elements to wedge automatically.

0° compares the angle to exactly zero so it can find spurious edges because of finite precision coordinates. Do you think it should instead use a small non-zero threshold and not allow exactly 0?

SergioP1975, 2024-05-30:
You are right, have checked and at the edge they are colapsed hexas instead of wedges. If the cuads should be converted to edges in the center of rotation, will be valid also for triangles lying in the center? Will be those converted to wedges, piramids or tets?

And for the thresshold 0°, the initial mesh lies in a plane, and all elements were revoluted by the same angle, so for the user the rear faces should lie also in a plane. Guess that you should relax internally a little such 0° tolerance

[image]

victorkemp, 2024-06-01:
I've added a tolerance of 1e-6 rad to the minimum threshold for edge detecting selection so that it picks up faces that you'd expect to be flat. Will be in v25.

closed 2024-06-01 by victorkemp