victorkemp/Mecway #44: Dynamic response 3d

sofien73, 2024-06-07:
I had some trouble with a beam model dynamic response analysis.
Seems that beam releases are not allowed with this type of analysis.
Is it normal?

victorkemp, 2024-06-09:
Yes, that's normal but is a feature that I might add later.

You can use constraint equations instead. Rather than releasing DOFs of elements sharing a node, couple the other DOFs on two separate nodes.

sofien73, 2024-07-19:
Victor, I try again on my previous claim within dynamic response on MECWAY website:

Hi every one, I've finally launched my first dynamic response analysis with a seismic spectra (T(s) vs Acceleration (ms-²)).
I have some trouble with beam release which it's not allowed with this kind of analysis but why is it not possible (maybe go back to constraint equation)?
Maybe for the next version of MW.

Another thing that could be better is having the possibility to extract maximum value of the outputs (beam forces and moment sorted by nodes).
The outputs is allowable by time step which it's normal the idea is too output maximum value by nodes in the .csv file.

One last thing is to have the possibility to disable some of the columns in my exemple I don't want to have the time step and the time in my csv file.

Regards.

victorkemp, 2025-03-25:
Not for the internal solver, but it is now available with nonlinear explicit dynamics and the OpenRadioss solver.

sofien73, 2025-03-25:
> Not for the internal solver, but it is now available with nonlinear explicit dynamics and the OpenRadioss solver.

Thanks Victor, I perform linear time history analysis in that case hope this will be updated soon with internal solver for the next version because it's really a problem to design a steel frame without beam releases in seismic conditions (modal specral) (I favor the constraint equation but it will not work properly for bracings)

sofien73, 2025-11-16:
Sorry Victor I have not pay enough attention to the previous message. OK for line pressure on beams but no contraint equation is impossible when I have several beams connected on the same node as we have seen before I need to connect beams on a Master node to extract Results properly. How can I deal with this problem ??