victorkemp/Mecway #59: Material Properties window is too narrow to show a complete line of OpenRadioss matedial definition

SergioP1975, 2025-01-05:
Hi, the Material Properties has a fixed size and cannot show a complete line of OpenRadioss definition. Also could include the header/footer to make it easy to write it in the needed position.

#---1----|----2----|----3----|----4----|----5----|----6----|----7----|----8----|----9----|---10----|

victorkemp, 2025-01-05:
I see this happens at certain display scaling settings like 125% and 225% but not at 100%, 150%, 175% or 200%. I'll fix it for version 28. Thanks.

SergioP1975, 2025-01-06:
Thanks! I have noted that cannot copy from Window Notepad and past directly in the Material Properties text field, the line breacks are not keeped. Weird thing, because the Notepad is the more basic text that I can imagine.

victorkemp, 2025-01-06:
Do you just mean the lines wrap like this?
[image]

or is it also ignoring line breaks like this? I can't reproduce this
[image]

SergioP1975, 2025-01-06:
This is copying and pasting directly from Notepad:

[image]

And this is pasting first in a "stickie note" (Stickies by Tom Revell), coping again and then pasting in the material window:

[image]

SergioP1975, 2025-01-06:
If you look in the lower side of the Notepad they say "UNIX (LF)", I have downloaded the files from th OpenRadioss examples, could if be that is taking a different charcater for the line break???? Maybe the Stickies application is doing the conversion

victorkemp, 2025-01-06:
I see, thanks. That's a broader problem with all multiline text inputs. I'll make a new issue for it.

closed 2025-01-10 by victorkemp