victorkemp/Mecway #61: The "input File" button on OpenRadioss Solver window doesn´t work

SergioP1975, 2025-01-06:
Hi, the input file button on the OpenRadioss solver window is not working, it trys to run or open a CCX related task. The temporal .rad file is created, but the button doesn´t open it.

[image]

victorkemp, 2025-01-06:
Oh, it's using the CCX .inp file editor specified in Tools -> Options -> CCX which you must have set to something CCX specific. I may need to add a separate option for OpenRadioss input file editor.

SergioP1975, 2025-01-06:
You are right, I had made a mistake in the CCX input file editor. Now I have fixed and it works.

[image]